Stephan3/dwc2-for-klipper-socket #2: Main GUI keeps refreshing 500 error.

guythnick, 2020-09-12:
I have just installed dwc2-for-klipper-socket via the KIAUH install script.  When I go to the main page, it keeps refreshing showing Operation failed (Reason: 500 error).  Basically 2-5 times per second. 

I have also installed Mainsail, and it is functioning fine.  I have also noticed that in dwc2, it does pull the device's data each time it refreshes.  

I also see nothing in the Klippy.log file that indicates an error.  The Klippy and dwc2 services are both running without error in systemd.  I have tried restarting my printer and raspberry pi in various different orders.  I have installed this from a fresh copy of raspberry pi OS lite on a pi 3b+. 

[klippy.log](https://github.com/Stephan3/dwc2-for-klipper-socket/files/5213112/klippy.log)
[printer.cfg.txt](https://github.com/Stephan3/dwc2-for-klipper-socket/files/5213115/printer.cfg.txt)

lenne0815, 2020-09-12:
I had the exact same issue today, but was still on the old dwc2 version. While i fiddled with it the script was updated here on github, i did a new git pull, installed dwc again and from then on it worked fine. I also flashed a fresh updated firmware to the mcu.

Stephan3, 2020-09-12:
dwc does not log to klippy log. I need to see whats going on. can you run it from console and show me the output?
there is no logfilesupport yet.

guythnick, 2020-09-12:
Yes, that did show a problem:

```ERROR:tornado.application:Uncaught exception GET /rr_config (192.168.1.229)
HTTPServerRequest(protocol='http', host='192.168.1.121:4750', method='GET', uri='/rr_config', version='HTTP/1.1', remote_ip='192.168.1.229')
Traceback (most recent call last):
  File "/home/pi/dwc-env/lib/python3.7/site-packages/tornado/web.py", line 1703, in _execute
    result = await result
  File "/home/pi/dwc2-for-klipper-socket/rr_handler.py", line 54, in get
    await rr_config(self)
  File "/home/pi/dwc2-for-klipper-socket/rr_handler.py", line 156, in rr_config
    float( self.poll_data['configfile']['config']['stepper_z']['position_endstop'] )
KeyError: 'position_endstop'
```

My z stepper config section below.  I don't have a position_endstop.  Using the same config I have in the past for Mainsail and Octoprint.  If I add 'position_endstop: 0', it throws the same error. 

```[stepper_z]
step_pin: PB0
dir_pin: PC5
enable_pin: !PB1
step_distance: .0025
endstop_pin:probe:z_virtual_endstop
position_max: 250
position_min: -8```

guythnick, 2020-09-12:
I commented that line out from rr_handler.py, and it appears to run now.  Not sure what the negative aspects are yet. 

```
		"axisMins": [ 	float( self.poll_data['configfile']['config']['stepper_x']['position_endstop'] ),
						float( self.poll_data['configfile']['config']['stepper_y']['position_endstop'] ),
						#float( self.poll_data['configfile']['config']['stepper_z']['position_endstop'] )
```

guythnick, 2020-09-12:
That was messing up z tracking, but I just hard-coded 0.0 and now it seems to be working. 

```
		"axisMins": [ 	float( self.poll_data['configfile']['config']['stepper_x']['position_endstop'] ),
						float( self.poll_data['configfile']['config']['stepper_y']['position_endstop'] ),
						0.0
		],
```

Stephan3, 2020-09-13:
@guythnick your config errors me btw
Error:  No option 'z_offset' in section: 'bltouch'

Stephan3, 2020-09-13:
should work now as intended. please checkout latest master.

guythnick, 2020-09-13:
Yes, it is now working.  Thank you!

closed 2020-09-13 by guythnick