Stephan3/dwc2-for-klipper-socket #4: Failed to get file list: Target drive is unmounted, Can't detect virtual sdcard

sealj553, 2020-09-13:
Hi,
I just installed this with th33xitus's installer, and it looks like everything is working as intended, but when I try to upload files, I get the following errors on the bottom left:
```
Failed to get file list: Target drive is unmounted
```
```
Can't detect virtual sdcard
```
I tried with and without virtual_sd enabled.
I did get it to print once by hitting `Upload & Start`, but it no longer works and I don't know what changed. I've never been able to upload files through the `Jobs` tab, though.

Relevant section of my `printer.cfg`:
```
[virtual_sdcard]
path: /home/pi/sdcard
```
This used to be an Octoprint install, but I gutted it. I don't think I left anything enabled that should interfere. I also went through the installation instructions after using the script and made sure everything is set up correct.
Let me know if there are any relevant logs I can include.
Thanks for all the hard work on this project!

Stephan3, 2020-09-13:
Hi!

Can sounds like an issue with your folder. Can you recheck that everything is fine with /home/pi/sdcard?
might be nonexistent? or just remove it and recreate as a pi user?
During startup the first few seconds not all klipperdata is present. The webif is throwing that error then but weverything works fine.

sealj553, 2020-09-13:
The `/home/pi/sdcard` directory does indeed exist. I've also tried to `chmod -R 777` and `chown -R pi:pi` it just in case, no luck. I've also tried  creating `/home/pi/jobs` directory for instance, with the same result.
I just confirmed my installation got extra borked, so I'm going to reinstall from scratch manually and see how it goes.

Stephan3, 2020-09-13:
try running it from console. enter the directory where your web_dwc2.py is located and rut it using python3 we_dwc2.py

sealj553, 2020-09-13:
Just reinstalled fresh. Not sure if this is the same issue, but:
```
pi@printer:~/dwc2-for-klipper-socket $ python3 web_dwc2.py
ERROR:tornado.application:Exception in callback functools.partial(<bound method IOLoop._discard_future_result of <tornado.platform.asyncio.AsyncIOMainLoop object at 0xb5d3beb0>>, <Task finished coro=<dwc2.init_() done, defined at web_dwc2.py:150> exception=TypeError('expected str, bytes or os.PathLike object, not NoneType')>)
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/tornado/ioloop.py", line 743, in _run_callback
    ret = callback()
  File "/home/pi/.local/lib/python3.7/site-packages/tornado/ioloop.py", line 767, in _discard_future_result
    future.result()
  File "web_dwc2.py", line 181, in init_
    self.sd_root = os.path.expanduser( configfile.get('config',{}).get('virtual_sdcard',{}).get('path', None) )
  File "/usr/lib/python3.7/posixpath.py", line 235, in expanduser
    path = os.fspath(path)
TypeError: expected str, bytes or os.PathLike object, not NoneType
```
The process doesn't exit, but there's no web interface.

Stephan3, 2020-09-13:
looks like your virtual sdcard is not recognized. give us. please login to console and give us the polldata:
curl http://127.0.0.1:4700/rr_poll_data

sealj553, 2020-09-13:
I ran `python3 web_dwc2.py` in another shell, then:
```
pi@printer:~ $ curl http://127.0.0.1:4700/rr_poll_data
curl: (7) Failed to connect to 127.0.0.1 port 4700: Connection refused
```
Looks like the server isn't running. Am I missing something?

Stephan3, 2020-09-14:
Wrong port sorry 4750

sealj553, 2020-09-15:
```
pi@printer:~ $ curl http://127.0.0.1:4750/rr_poll_data
{"last_path": null, "klipper_macros": [], "gcode/help": {"STATUS": "Report the printer status", "SAVE_CONFIG": "Overwrite config file and restart", "RESTART": "Reload config file and restart host software", "FIRMWARE_RESTART": "Restart firmware, host, and reload config"}, "info": {"state_message": "Unable to open config file /home/pi/printer.cfg\nOnce the underlying issue is corrected, use the \"RESTART\"\ncommand to reload the config and restart the host software.\nPrinter is halted\n", "klipper_path": "/home/pi/klipper", "config_file": "/home/pi/printer.cfg", "software_version": "v0.8.0-786-g063f9a2c", "hostname": "printer", "cpu_info": "4 core ARMv7 Processor rev 3 (v7l)", "state": "error", "python_path": "/home/pi/klippy-env/bin/python", "log_file": "/tmp/klippy.log"}, "objects/list": {"objects": ["webhooks", "configfile"]}, "list_endpoints": {"endpoints": ["info", "list_endpoints", "emergency_stop", "gcode/restart", "gcode/subscribe_output", "objects/query", "objects/list", "gcode/firmware_restart", "objects/subscribe", "gcode/script", "gcode/help"]}, "gcode/subscribe_output": {}, "webhooks": {"state": "error", "state_message": "Unable to open config file /home/pi/printer.cfg\nOnce the underlying issue is corrected, use the \"RESTART\"\ncommand to reload the config and restart the host software.\nPrinter is halted\n"}, "configfile": {"config": {}}}
```
Now this problem is much clearer: `Unable to open config file /home/pi/printer.cfg`
Dumb error on my part, I forgot to copy over my config.

Now, it looks like it's working under the hood:
```
pi@printer:~/dwc2-for-klipper-socket $ curl localhost:4750/rr_poll_data
{"last_path": null, "klipper_macros": ["G29"], "gcode/help": {"SDCARD_RESET_FILE": "Clears a loaded SD File. Stops the print if necessary", "RESTORE_GCODE_STATE": "Restore a previously saved G-Code state", "BLTOUCH_STORE": "Store an output mode in the BLTouch EEPROM", "QUERY_ADC": "Report the last value of an analog pin", "TUNING_TOWER": "Tool to adjust a parameter at each Z height", "SAVE_CONFIG": "Overwrite config file and restart", "SET_DISPLAY_GROUP": "Set the active display group", "G29": "G-Code macro", "SAVE_GCODE_STATE": "Save G-Code coordinate state", "SET_PRESSURE_ADVANCE": "Set pressure advance parameters", "SET_GCODE_OFFSET": "Set a virtual offset to g-code positions", "PID_CALIBRATE": "Run PID calibration test", "INIT_TMC": "Initialize TMC stepper driver registers", "QUERY_PROBE": "Return the status of the z-probe", "BED_MESH_OUTPUT": "Retrieve interpolated grid of probed z-points", "TURN_OFF_HEATERS": "Turn off all heaters", "SET_STEPPER_ENABLE": "Enable/disable individual stepper by name", "QUERY_ENDSTOPS": "Report on the status of each endstop", "SET_TMC_CURRENT": "Set the current of a TMC driver", "Z_ENDSTOP_CALIBRATE": "Calibrate a Z endstop", "MANUAL_PROBE": "Start manual probe helper script", "DUMP_TMC": "Read and display TMC stepper driver registers", "PROBE_ACCURACY": "Probe Z-height accuracy at current XY position", "STATUS": "Report the printer status", "SDCARD_PRINT_FILE": "Loads a SD file and starts the print.  May include files in subdirectories.", "BLTOUCH_DEBUG": "Send a command to the bltouch for debugging", "SET_TMC_FIELD": "Set a register field of a TMC driver", "FIRMWARE_RESTART": "Restart firmware, host, and reload config", "STEPPER_BUZZ": "Oscillate a given stepper to help id it", "BED_MESH_CLEAR": "Clear the Mesh so no z-adjusment is made", "SET_GCODE_VARIABLE": "Set the value of a G-Code macro variable", "SET_VELOCITY_LIMIT": "Set printer velocity limits", "SET_PIN": "Set the value of an output pin", "SET_HEATER_TEMPERATURE": "Sets a heater temperature", "BED_MESH_PROFILE": "Bed Mesh Persistent Storage management", "PROBE": "Probe Z-height at current XY position", "BED_MESH_MAP": "Serialize mesh and output to terminal", "SET_EXTRUDER_STEP_DISTANCE": "Set extruder step distance", "PROBE_CALIBRATE": "Calibrate the probe's z_offset", "BED_MESH_CALIBRATE": "Perform Mesh Bed Leveling", "RESTART": "Reload config file and restart host software", "ACTIVATE_EXTRUDER": "Change the active extruder"}, "info": {"state_message": "Section 'web_dwc2' is not a valid config section\nOnce the underlying issue is corrected, use the \"RESTART\"\ncommand to reload the config and restart the host software.\nPrinter is halted\n", "klipper_path": "/home/pi/klipper", "config_file": "/home/pi/printer.cfg", "software_version": "v0.8.0-786-g063f9a2c", "hostname": "printer", "cpu_info": "4 core ARMv7 Processor rev 3 (v7l)", "state": "error", "python_path": "/home/pi/klippy-env/bin/python", "log_file": "/tmp/klippy.log"}, "objects/list": {"objects": ["webhooks", "configfile", "probe", "gcode_move", "bed_mesh", "heaters", "heater_bed", "heater_fan nozzle_cooling_fan", "fan", "menu", "display_status", "output_pin beeper", "gcode_macro G29", "print_stats", "virtual_sdcard", "query_endstops", "idle_timeout", "toolhead", "extruder"]}, "list_endpoints": {"endpoints": ["info", "list_endpoints", "query_endstops/status", "emergency_stop", "gcode/restart", "gcode/subscribe_output", "objects/query", "objects/list", "gcode/firmware_restart", "objects/subscribe", "gcode/script", "gcode/help"]}, "gcode/subscribe_output": {}, "heaters": {"available_sensors": ["heater_bed", "extruder"], "available_heaters": ["heater_bed", "extruder"]}, "heater_fan nozzle_cooling_fan": {"speed": 0.0}, "bed_mesh": {"mesh_max": [0.0, 0.0], "mesh_matrix": [ [... cut ...] section\nOnce the underlying issue is corrected, use the \"RESTART\"\ncommand to reload the config and restart the host software.\nPrinter is halted\n"}, "virtual_sdcard": {"progress": 0.0, "is_active": false, "file_position": 0}, "heater_bed": {"temperature": 0.0, "target": 0.0}, "extruder": {"pressure_advance": 0.05, "target": 0.0, "temperature": 0.0, "smooth_time": 0.02}}
```

Now it's running without error. Though when I try to `http://192.168.1.164:4750/`, I get `500: Internal Server Error`

Stephan3, 2020-09-15:
@sealj553 as stated above, run from console, i cant guess whats on.

sealj553, 2020-09-15:
Right, my bad.
I unplugged my printer, restarted the klipper service, and tried again. Now the interface is working:

```
pi@printer:~/dwc2-for-klipper-socket $ python3 web_dwc2.py
DWC2:.1510239 - 192.168.1.149 - GET - /
DWC2:.3936486 - 192.168.1.149 - GET - /css/app.c3eac487.css
DWC2:.7278302 - 192.168.1.149 - GET - /js/app.289e2336.js
DWC2:.2660468 - 192.168.1.149 - GET - /rr_connect?password=reprap&time=2020-9-15T7%3A47%3A21
DWC2:.3660364 - 192.168.1.149 - GET - /fonts/materialdesignicons-webfont.a323c28e.woff2
DWC2:.4143388 - 192.168.1.149 - GET - /rr_download?name=0%3A%2Fsys%2Fdwc2settings.json
DWC2:.4651139 - 192.168.1.149 - GET - /rr_filelist?dir=0%3A%2Fmacros&first=0
DWC2:.4829457 - 192.168.1.149 - GET - /rr_download?name=0%3A%2Fsys%2Fdwc2defaults.json
DWC2:.7067943 - 192.168.1.149 - GET - /rr_filelist?dir=0%3A%2Fmacros&first=0
```
The logging is working by the way.

It's showing status: off, no axes, but looks like it's connected to klipper.
[image]

Stephan3, 2020-09-15:
What is the result of status?
Enter status as gcode

sealj553, 2020-09-15:
Kind of a dumb error again. It there was a syntax error in my klipper config because I had added a `[web_dwc2]` block.
Removing that and restarting klipper fixed that. It can now connect. I can also upload gcode files now. I have a print running now.
Though I am getting a lot of 500 errors, with the interface flashing between "Connection established" and "Connection interrupted..." rendering it unusable. Also "Failed to get file info" when I uploaded a gcode file.

Here's the log: https://pastebin.com/Lx4cziHQ

Stephan3, 2020-09-15:
can you provide that gcode to me please?

top_front_.gcode

sealj553, 2020-09-15:
Had to add extension so github would let me upload: [top_front_.gcode.txt](https://github.com/Stephan3/dwc2-for-klipper-socket/files/5228553/top_front_.gcode.txt)

Stephan3, 2020-09-16:
please check again latest master. feels safer now.

sealj553, 2020-09-16:
It's working great now, thanks! Just started a print a few times to test, no errors at all.
I'll see if I can help with some documentation :)

closed 2020-09-16 by sealj553

Alexdad76, 2020-09-19:
The same situation, I cannot connect to the printer (th33xitus's installer).
Status - Off 

[dwc2.log](https://github.com/Stephan3/dwc2-for-klipper-socket/files/5250087/dwc2.log)
[klippy.log](https://github.com/Stephan3/dwc2-for-klipper-socket/files/5250088/klippy.log)

Alexdad76, 2020-09-20:
> 
> 
> The same situation, I cannot connect to the printer (th33xitus's installer).
> Status - Off
> 
> [dwc2.log](https://github.com/Stephan3/dwc2-for-klipper-socket/files/5250087/dwc2.log)
> [klippy.log](https://github.com/Stephan3/dwc2-for-klipper-socket/files/5250088/klippy.log)
> 
> PS: Before that, Octoprint worked fine.
> 
> System:
> 
> > Welcome to Ubuntu Xenial with Armbian Linux 3.4.113-sun8i
> > System load:   0.27 0.31 0.27   Up time:       1:00 hour
> > Memory usage:  27 % of 1000MB   IP:            192.168.1.67
> > CPU temp:      23°C
> > Usage of /:    32% of 15G
> 
>     * SKR 1.1
> 
> 
> PS1 I'll try to reinstall another system based on Debian.

Armbian_20.08.1_Orangepipcplus_buster_current_5.8.5.img.xz  works but keeps giving error : HTTP request timed out

Alexdad76, 2020-09-20:
> Armbian_20.08.1_Orangepipcplus_buster_current_5.8.5.img.xz works but keeps giving error : HTTP request timed out

After some time, the error no longer appears.