Stephan3/dwc2-for-klipper-socket #5: Cura gcode parsing not working quire right

guythnick, 2020-09-13:
Using Cura 4.6.2, it appears the parse_gcode is not working 100%.  The filament variable is reading in meters, but storing (or at least, the display in web interface) is in millimeters.  I think multiplying by 10 in the parse code should work.

I am also having trouble with layer calculations with many of my jobs, but I can't quite put my finger on why yet.  I think the estimated time is also not parsing correctly. Perhaps it will be apparent comparing the code to the actual file output below.  

Here is the problem portion of code: 

https://github.com/Stephan3/dwc2-for-klipper-socket/blob/ffd35c7e0d43120bbc4a0d5035a9360111d6e4e2/rr_handler.py#L798-L808

Here is what is in the header output in a CURA file:

```
;FLAVOR:Marlin
;TIME:1341
;Filament used: 0.564835m
;Layer height: 0.2
;MINX:107.7
;MINY:107.7
;MINZ:0.2
;MAXX:127.3
;MAXY:127.3
;MAXZ:20
```

My list of jobs, you can see the data is either not there or incorrect:
[image]

Here is a screenshot of one of my jobs that is not showing layers correctly (there are 12 layers in this one):

[image]

Also including the gcode from the screenshot above:

[Greek_Meander_Lamp_part22.gcode.txt](https://github.com/Stephan3/dwc2-for-klipper-socket/files/5214822/Greek_Meander_Lamp_part22.gcode.txt)

Stephan3, 2020-09-13:
@th33xitus this one is yours :)

i look into that

th33xitus, 2020-09-13:
@Stephan3 I haven't touched the Cura section in my last PR. But i saw that issue too. I would have fixed it but the filament is, like already mentioned, noted in meters and not in millimeters like in almost every other slicer (because Cura needs to be special right?!). So we would need a conversion step there, nothing you can do with simple regex, or?

What we could do is, grab the `MAXZ:` parameter for reading the max height of the object instead of looking for the last `G1 Z...` command like it's done now.

lenne0815, 2020-09-13:
Sorry to interrupt but i just stumbled across a problem, for me only cura works at all, s3d and prusaslicer both give "slicer not supported". As far as i can tell its because the additional stuff here 'name': 'PrusaSlicer\s\d+.\d+.\d+', and here 'name': 'Simplify3D.*\s\d+.\d+.\d+', compared to 'name': 'Cura_SteamEngine.*',

th33xitus, 2020-09-13:
@lenne0815 Can you provide example g-code files which don't work for you?
Before i did the PR i made sure it works... well... on my setup at least. You can see it in the screenshot i provided with the PR here: https://github.com/Stephan3/dwc2-for-klipper-socket/pull/1

lenne0815, 2020-09-13:
Goddamnit, i just realized i installed it without your latest patch, it works now, sorry ! Huge thanks for the work from both of you !

guythnick, 2020-09-13:
I think it will work if you use this.  In the middle of a print, so can't test it now.  But, the regex works at least for all of them. For TIME I just removed the extra "\".

```
		'Cura':
			{
				'name': 'Cura_SteamEngine.*',								#	somewhere in the first lines
				'object_h': ';MAXZ:\d*.\d',									#	get the highest knowing z
				'first_h': ';MINZ:\d*.\d',									#	get the lowest knowing z
				'layer_h': ';Layer height: \d.\d+',							#	its there
				'duration': ';TIME:\d+',									#	its there
				'filament': ';Filament used: \d*.\d+m'						#	its there
			},
```

commit 789a8bc8 (2020-09-14): Overhaul all slicer regexes. fixes #5
closed 2020-09-14 by Stephan3