Stephan3/dwc2-for-klipper-socket #6: reconnect

jornada812, 2020-09-14:
I used the kiauh script. It downloads the socket version as I understand it. The reconnection error persists.

Stephan3, 2020-09-14:
@jornada812 
can you run it from console and see what errors? We need to identify whats up in your case.

jornada812, 2020-09-14:
sorry i'm a windows user) what should i do to run from console?

jornada812, 2020-09-14:
in addition, I can say that after installing KIAUH I do not have any services here /etc/systemd/system/klipper.service and here /etc/systemd/system/dwc.service (should they be?) All I did was install it clean raspberry pi os through imager set up ssh, wifi and run the script.

lenne0815, 2020-09-14:
No, its empty for me aswell and everything works here. Install WinSCP go to /home/pi open klippy.log and check if klipper throws any config errors ( is the config.cfg even correctly copied to /home/pi/ ? ), that happend for me on first install.

jornada812, 2020-09-14:
the configuration is correct. no mistakes. in the console i get this all the time
Connection interrupted, attempting to reconnect ...
Operation failed (Reason: 500: Internal Server Error)
14.09.2020, 15:47:24 Connection established
09/14/2020, 3:47:24 PM Connection interrupted, attempting to reconnect ...
Operation failed (Reason: 500: Internal Server Error)
14.09.2020, 15:47:23 Connection established

I wrote about the presence of services, because the installation description says - "This s my klipper systemd service located at /etc/systemd/system/klipper.service"

jornada812, 2020-09-14:
Maybe all this works with a specific version of dwc and not the latest? if so, what should i do to get it working?

Stephan3, 2020-09-14:
@jornada812 it works with latest klipper and latest dwc. 
Hard to help you. i assume the installerscript puts dwc translator to /home/pi

so lets do it like this:
```
systemctl stop dwc dwc2
/etc/init.d/dwc stop
su pi
cd /home/pi/dwc*socket
python3 web_dwc2.py
```
then connect to the webinterface, close it after err 500 occurs and post the output here.

jornada812, 2020-09-14:
something like this (I really don't have such a section stepper_x in the config)
DWC2:.2616756 - 127.0.0.1 - GET - /rr_connect?password=reprap&time=2020-9-14T17%3A51%3A46
ERROR:tornado.application:Uncaught exception GET /rr_config (127.0.0.1)
HTTPServerRequest(protocol='http', host='192.168.1.65', method='GET', uri='/rr_config', version='HTTP/1.1', remote_ip='127.0.0.1')
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/tornado/web.py", line 1592, in _execute
    result = yield result
  File "/usr/lib/python3/dist-packages/tornado/gen.py", line 1133, in run
    value = future.result()
  File "/home/pi/dwc2-for-klipper-socket/rr_handler.py", line 54, in get
    await rr_config(self)
  File "/home/pi/dwc2-for-klipper-socket/rr_handler.py", line 153, in rr_config
    x = self.poll_data['configfile']['config']['stepper_x']
KeyError: 'stepper_x'
DWC2:.3973985 - 127.0.0.1 - GET - /rr_connect?password=reprap&time=2020-9-14T17%3A51%3A47
ERROR:tornado.application:Uncaught exception GET /rr_config (127.0.0.1)
HTTPServerRequest(protocol='http', host='192.168.1.65', method='GET', uri='/rr_config', version='HTTP/1.1', remote_ip='127.0.0.1')
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/tornado/web.py", line 1592, in _execute
    result = yield result
  File "/usr/lib/python3/dist-packages/tornado/gen.py", line 1133, in run
    value = future.result()
  File "/home/pi/dwc2-for-klipper-socket/rr_handler.py", line 54, in get
    await rr_config(self)
  File "/home/pi/dwc2-for-klipper-socket/rr_handler.py", line 153, in rr_config
    x = self.poll_data['configfile']['config']['stepper_x']
KeyError: 'stepper_x'
DWC2:.3452067 - 127.0.0.1 - GET - /rr_connect?password=reprap&time=2020-9-14T17%3A51%3A48
ERROR:tornado.application:Uncaught exception GET /rr_config (127.0.0.1)
HTTPServerRequest(protocol='http', host='192.168.1.65', method='GET', uri='/rr_config', version='HTTP/1.1', remote_ip='127.0.0.1')
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/tornado/web.py", line 1592, in _execute
    result = yield result
  File "/usr/lib/python3/dist-packages/tornado/gen.py", line 1133, in run
    value = future.result()
  File "/home/pi/dwc2-for-klipper-socket/rr_handler.py", line 54, in get
    await rr_config(self)
  File "/home/pi/dwc2-for-klipper-socket/rr_handler.py", line 153, in rr_config
    x = self.poll_data['configfile']['config']['stepper_x']
KeyError: 'stepper_x'
DWC2:.3507016 - 127.0.0.1 - GET - /rr_connect?password=reprap&time=2020-9-14T17%3A51%3A49
ERROR:tornado.application:Uncaught exception GET /rr_config (127.0.0.1)
HTTPServerRequest(protocol='http', host='192.168.1.65', method='GET', uri='/rr_config', version='HTTP/1.1', remote_ip='127.0.0.1')
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/tornado/web.py", line 1592, in _execute
    result = yield result
  File "/usr/lib/python3/dist-packages/tornado/gen.py", line 1133, in run
    value = future.result()
  File "/home/pi/dwc2-for-klipper-socket/rr_handler.py", line 54, in get
    await rr_config(self)
  File "/home/pi/dwc2-for-klipper-socket/rr_handler.py", line 153, in rr_config
    x = self.poll_data['configfile']['config']['stepper_x']
KeyError: 'stepper_x'

jornada812, 2020-09-14:
Probably for the delta printer rr_handler.py should be slightly different?

jornada812, 2020-09-14:
i tried to replace with
x = self.poll_data ['configfile'] ['config'] ['stepper_a']
y = self.poll_data ['configfile'] ['config'] ['stepper_b']
z = self.poll_data ['configfile'] ['config'] ['stepper_c']
but now KeyError is raised: 'position_max'
but everything is not the same as on the сartesians and I do not know what to write

Stephan3, 2020-09-14:
you are on the right track, give me your config please.

jornada812, 2020-09-14:
https://yadi.sk/d/wC48SEqbGGcb0g

Stephan3, 2020-09-15:
@jornada812 please try latest master now. if everything works.

jornada812, 2020-09-15:
Nearly) When trying any movement using interface buttons :
M120 G91 G1 Y-50 F6000 G90 M121
Could not run code because a bad response has been received

Error:  Move out of range: 1.802 1.420 486.536 [0.000]
M120 G91 G1 X1 F6000 G90 M121
Could not run code because a bad response has been received

jornada812, 2020-09-15:
Reconnections disappeared. The primary question is closed, but problems remain. Make a new request?

Stephan3, 2020-09-15:
yes

closed 2020-09-15 by Stephan3