Stephan3/dwc2-for-klipper-socket #7: few questions

jornada812, 2020-09-15:
As far as I remember, the duet had 3 calculation times. Based on the use of filament, only one of them. Are the rest unfulfilled?
Why does not it show the number of the layer that is printing?
If you manually set the fan speed, and in the slicer on the next layer a speed will be set that differs from the one set in the interface, then the fan speed will change as indicated by the slicer. For this case, the original interface of the duo provided fixing buttons that returned the speed as in the interface. This is also not the case.
And finally, RRF G32 teams and others. Is it possible to replace with clipper commands?

Stephan3, 2020-09-15:
> As far as I remember, the duet had 3 calculation times. Based on the use of filament, only one of them. Are the rest unfulfilled?
Why does not it show the number of the layer that is printing?

- we do have based on slicer/filprogress/filamentusage
- layertimes is outstanding, its more complicated to get. Take a look at that code from earlier version, then you get what i mean. I want to give it an overhaul later. 

> If you manually set the fan speed, and in the slicer on the next layer a speed will be set that differs from the one set in the interface, then the fan speed will change as indicated by the slicer.
- we trickled that down weeks ago. It has to do with DWC see
https://github.com/Duet3D/DuetWebControl/issues/269

>  the original interface of the duo provided fixing buttons
no sure waht you mean, never had a duet.

> And finally, RRF G32 teams and others. Is it possible to replace with clipper commands?
- thats what i do, which one did ai miss see:
https://github.com/Stephan3/dwc2-for-klipper-socket/blob/f2b1c329c657ce06ed13b36a0a8141bd449d3b8c/rr_handler.py#L322

jornada812, 2020-09-16:
It looked like this - if you put a checkbox next to the selected fan value, then if the fan speed is changed, priority is given to the interface, if not, then the slicer.
About G32: menu Compensation & calibration - is it possible to replace menu items there with clipper commands?

Stephan3, 2020-09-16:
> It looked like this - if you put a checkbox next to the selected fan value, then if the fan speed is changed, priority is given to the interface, if not, then the slicer.

My code is independent from the webinterface. No idea whats triggering the button to apear.

> About G32: menu Compensation & calibration - is it possible to replace menu items there with clipper commands?
If i would do this a bed_mesh_calibrate or quad_gantry_level take place every time a user is hitting "start file for print".
For me thats something that never happen. So if you want your bed meshed or delta calibrated use a startgcode of choice or override gcodes by making macros.

Stephan3, 2020-09-17:
@jornada812 I took a look at Duetwebcontrols code and still not sure what you mean.
please take a look to this ticket.
https://github.com/Duet3D/DuetWebControl/issues/269

jornada812, 2020-09-17:
> Duetwebcontrols code and still not sure what you mean. Screenshot?
> Duet3D/DuetWebControl#269

https://yadi.sk/i/J3pBd1dKBHSX5Q
I don't use duet anymore. There are pins in this picture. If they are
pressed, the fan speed set in the interface is supported, regardless of the
speed in gcode

Stephan3, 2020-09-18:
Bro that is duet webcontrol 1
That has been rewritten

closed 2020-09-27 by Stephan3