Stephan3/dwc2-for-klipper-socket #11: http request timout after print start

Matou25, 2020-10-07:
Hi,

When i start a new print it's impossible to do anything after:
changing temperature
pause/cancel job
the only way is to press emergency button.
is it normal ?

I use last version of Klipper (host and firmware) and dwc2 freshly install on a new raspbian lite OS?

thx

Stephan3, 2020-10-07:
The discribed behaviour does not happen for me. 
can you provide the dwc logfile dron /tmp ?

Matou25, 2020-10-07:
Tornado:.5925019 - 10.10.10.100 - GET - /rr_filelist?dir=0%3A%2Fgcodes&first=0
Tornado:.0645847 - 10.10.10.100 - GET - /rr_gcode?gcode=M32%20%220%3A%2Fgcodes%2F3d64e3e3-c6d8-463b-a522-5f42655f327a.gcode%22
Tornado:.6586182 - 10.10.10.100 - GET - /rr_fileinfo
timeout reached with: Timeout
Tornado:.2376304 - 10.10.10.100 - GET - /rr_gcode?gcode=M25
timeout reached with: Timeout
Tornado:.7544584 - 10.10.10.100 - GET - /rr_gcode?gcode=M0%20H1
Tornado:.2437897 - 10.10.10.100 - GET - /rr_gcode?gcode=M0%20H1
Tornado:.4864118 - 10.10.10.100 - GET - /rr_gcode?gcode=M24

there is time out if you try to pause/abord print or change temperature when printer heat with M190/M109 (set temp and wait to reach it)

Stephan3, 2020-10-07:
@Matou25 Thats how klipper works. It does not process data once temp and wait for it is active.

Matou25, 2020-10-08:
@Stephan3 strange because before update ( klipper 0.7 and dwc2 for clipper), with the same stl file it work .
but it's probably du to le klipper firmware update i will ask in Klipper project, it's strange to not be able to stop printing during this waiting time.

thanks for your clear answer.

closed 2020-10-08 by Matou25