Stephan3/dwc2-for-klipper-socket #17: Implement IceSL slicer

1walkerit, 2020-10-26:
Hi, 
is it possible to implement IceSL slicer?
I have these at my header:
;===================
**;TOTAL INFORMATION**
**;Printer:** wscubexy-06tryska
**;Slicer:** IceSL version: 2.3.5
**;Object height:** 7.2
**;Number of layers:** 30
**;Total filament lenght:** 18399.701 mm
**;Print time:**   143.433 minutes
**;Print dimension:** X:89.498 mm Y:30.226 mm X1:242.503 mm Y1:149.774 mm Z:7.200 mm
;===================

Using older version of dwc2 I've hade working regex but now I am not possble to make it working.
Thank you very much

Stephan3, 2020-10-26:
please provide the complete gcode file.

1walkerit, 2020-10-26:
> please provide the complete gcode file.

I'm sending sample g-code
[[a]_upper_idler_support_a-b_right-ASA-06tr.gcode.zip](https://github.com/Stephan3/dwc2-for-klipper-socket/files/5438649/a._upper_idler_support_a-b_right-ASA-06tr.gcode.zip)