Stephan3/dwc2-for-klipper-socket #18: missing files with kiauh install

robthide37, 2020-11-15:
This is the error i get at the end shoud i delete and reinstall as init.d instead of system ?
###### Installing system start script...
Created symlink /etc/systemd/system/multi-user.target.wants/dwc.service → /etc/systemd/system/dwc.service.

###### Launching dwc2-for-klipper-socket...
>>>>>> Service installed!

###### Checking /etc/default/klipper for necessary entries ...
grep: /etc/default/klipper: No such file or directory

###### Patching unix domain socket to /etc/default/klipper ...
sed: can't read /etc/default/klipper: No such file or directory

Stephan3, 2021-01-29:
@th33xitus can you comment on that ?

th33xitus, 2021-01-29:
I think there might be a bug when KIAUH checks for the klipper uds argument being present in the service file. And it seems like it checks for the init.d service files, but the user has klipper installes as systemd.

I have no intereset in fixing this bug tbh, since i am reworking KIAUH and make every service only installable via systemd anyways, and make it obsolete to "patch" an argument to that service file.

Stephan3, 2021-01-29:
okay sounds reasonable. the next version will move forward. Thanks.

closed 2021-01-29 by Stephan3