Stephan3/dwc2-for-klipper-socket #21: Failed to get file list can't detect virtual sd card.

wormiedk, 2021-01-20:
I get the following in /tmp/dwc2.log
`
========== Started ==========
ERROR:tornado.application:Exception in callback functools.partial(<bound method IOLoop._discard_future_result of <tornado.platform.asyncio.AsyncIOMainLoop ob
ject at 0x760711f0>>, <Task finished coro=<read_stream() done, defined at /home/pi/dwc2-for-klipper-socket/web_dwc2.py:174> exception=TypeError("the JSON obj
ect must be str, not 'bytes'",)>)
Traceback (most recent call last):
  File "/home/pi/dwc-env/lib/python3.5/site-packages/tornado/ioloop.py", line 743, in _run_callback
    ret = callback()
  File "/home/pi/dwc-env/lib/python3.5/site-packages/tornado/ioloop.py", line 767, in _discard_future_result
    future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 293, in result
    raise self._exception
  File "/usr/lib/python3.5/asyncio/tasks.py", line 239, in _step
    result = coro.send(None)
  File "/home/pi/dwc2-for-klipper-socket/web_dwc2.py", line 181, in read_stream
    out_ = json.loads(data[:-1])
  File "/usr/lib/python3.5/json/__init__.py", line 312, in loads
    s.__class__.__name__))
TypeError: the JSON object must be str, not 'bytes'
`

Stephan3, 2021-01-20:
By doing what?

wormiedk, 2021-01-20:
When launching DWC. The web interface hangs with the "failed to get file list" and "can't detect virtual sd card" messages

Stephan3, 2021-01-20:
show us your klippy.log and dwc.log . we cant "guess"

Apar7777, 2021-01-29:
Hello
Same issue for me
Klipper and dcw are launched by systemd

 Added -a /tmp/klippy_uds in klipper.service

 Added in config: (is that correct ?)
[virtual_sdcard]
path = /home/pi/sdcard

Klipper run well with octoprint (octopi distrib)

[image]

[dwc2.log](https://github.com/Stephan3/dwc2-for-klipper-socket/files/5890758/dwc2.log)

[klippy.log](https://github.com/Stephan3/dwc2-for-klipper-socket/files/5890759/klippy.log)

Van-Hellsing, 2021-01-29:
Check the printer.cfg file it may be empty

Stephan3, 2021-01-29:
> Hello
> Same issue for me
> Klipper and dcw are launched by systemd
>
> Added -a /tmp/klippy_uds in klipper.service
>
> Added in config: (is that correct ?)
> [virtual_sdcard]
> path = /home/pi/sdcard
>
> Klipper [...]

looks correct to me. I il ltry your configuration locally.

adq, 2021-04-04:
Hey, maybe the original commented didn't realise you have to setup the virtual sd card in *klipper* config not in dwc.cfg. 

Had the same problem and adding this sorted it completely:
```
[virtual_sdcard]
path: /home/pi/sdcard/
```