Stephan3/dwc2-for-klipper-socket #24: g28 http request timed out

shawnp30, 2021-05-16:
hello. trying out klipper and dwc. i get this error when homing. "g28 http request timed out" everything works but it shows that error. i used kiauh to install for gtr v1.0 board. what can i change to fix the error?
[image]

ForbiddenEra, 2021-07-26:
So, I'm not sure if we can somehow respond to the HTTP faster from the connector or something, but I was able to modify the duet source to remove the error.

In 3.3.0 current, in /duetwebcontrol/js/app.24acd1b2.js on line 42, col 408428, change J.timeout to =25000 .. not ideal editing packed source, but a quick fix without recompiling .. also I changed rr_handler.py line 354 to wait 30..

Stephan3, 2021-07-26:
Its related to internal mechanic differences Klipper <> rrf
It cant be fixed without loosing the ability recieving exceptions after a OK